stm32-rs/meta #7: HAL: More precise units for timers.

agausmann, 2021-09-04:
Timers are currently implemented with `type Time = Hertz`. Besides the obvious oddity of using frequency as the unit for timeout times, the `Hertz` wrapper type is also limited to integer values, which severely restricts the possible periods/frequencies that can be set for a timer.

This isn't a problem at the kHz or MHz scale, but at low audible frequencies (like the low end of the [range of MIDI music notes](https://www.inspiredacoustics.com/en/MIDI_note_numbers_and_center_frequencies)), it's just not enough precision. At the very lowest values, the differences between frequencies are <1Hz, and above those, the frequencies are still noticeably out of tune until they reach a scale where integers provide enough precision.

agausmann, 2021-09-04:
As a counterexample, the `atsamd-hal` crate implements [their timer-counters](https://docs.rs/atsamd-hal/0.12.0/atsamd_hal/samd21/timer/struct.TimerCounter.html) using a `Nanoseconds` type.

Sh3Rm4n, 2021-09-06:
I've used the [`type Time = embedded_time::duration::Generic<u32>`](https://docs.rs/stm32f3xx-hal/0.8.0/stm32f3xx_hal/timer/struct.Timer.html#impl-CountDown) approach in [`stm32f3xx-hal`](https://github.com/stm32-rs/stm32f3xx-hal/blob/ca538b50fe4734c0358c0ac6d2f698a28ab33604/src/timer.rs#L238-L242), which might not be the most efficent solution (size wise as value and fraction has to be saved separately in a struct) but results into nice to use [APIs ](https://github.com/stm32-rs/stm32f3xx-hal/blob/ca538b50fe4734c0358c0ac6d2f698a28ab33604/testsuite/tests/timer.rs#L120) IMO, where you also have the more freedom about the [precision](https://github.com/stm32-rs/stm32f3xx-hal/blob/ca538b50fe4734c0358c0ac6d2f698a28ab33604/testsuite/tests/timer.rs#L80-L87). 

But I think, for use cases like yours, where the frequency has to be very accurate so they error margin are unaudible, I don't know if there is a nice API to abstract this or if the timer implementations have to expose their scaler values (PSC and ARR as they are abreviated for stm32) directly to fine tune the values