stm32-rs/meta #9: Merging stm32-rs HALs into one unified STM32 HAL

FeldrinH, 2025-07-15:
I think it would be valuable to have a unified stm32-rs HAL for all STM32 variants, similar to how esp32-rs and Embassy have a single HAL.

An annoyance that I've encountered with the current set of HAL-s is that different HAL-s seem to have slightly different API-s for various peripherals and this leads to friction when working with multiple HAL-s. Also the quality and completeness of HAL-s varies a lot. A unified HAL would help a lot to solve these pain points.

It might also help ease the maintenance burden. I figure having the common parts shared and a common API established would reduce duplicate effort accross HAL-s and help speed up the process of adding support for new STM32 device families.

Related discussions: https://github.com/rust-embedded/embedded-hal/issues/61, [stm32-rs matrix chat](https://matrix.to/#/!WqsLCItsZbJGhRjnxP:matrix.org/$e1TvQ6mXpig4jKyKpbY_w6jKD7b-m_A0mZOHgBYsITY?via=matrix.org&via=psion.agg.io&via=mozilla.org).

If there is interest in working towards merging the HALs from the maintainers of individual HALs, then I hope this issue can serve as a common place for working towards a unified HAL. I myself am open to contributing to developing a unified HAL, though unfortunately I don't have time to be a long-term maintainer of the project.

I would love to hear the thoughts of current HAL maintainers on this proposal (@burrbull, @astapleton, @usbalbin, @newAM, @therealprof).

newAM, 2025-07-16:
I think it's a good idea, but it's a lot of work.  Structurally I think a unified HAL would need to be in a single repo, similar to how embassy-stm32 does it.  As far as I know the embassy-stm32 HAL can be used without embassy; what would a unified HAL offer that isn't offered by embassy-stm32 currently?

FeldrinH, 2025-07-16:
> As far as I know the embassy-stm32 HAL can be used without embassy; what would a unified HAL offer that isn't offered by embassy-stm32 currently?

A valid question, but I think this question is equally valid for the current HALs. What do the current stm32-rs HALs offer that embassy-stm32 doesn't?

In my mind the main goal of a unified HAL is reducing ecosystem fragmentation. If embassy-stm32 already fills the role of a unified HAL then a more radical possibility is to just deprecate the stm32-rs HALs and direct maintenance resources to embassy-stm32. I don't know if that is feasible in practice, but it seems like a valid alternative in theory.

nisembedded, 2025-07-16:
I don't see any cons about this. It will better for consistency and for support new series of stm32 which have no many maintainers yet. Like stm32c0xx, u5, etc. Also will be better to see option like nightly features and use the night version of stm32 PAL, or even stop using it anymore.

therealprof, 2025-07-16:
I'm not opposed to this idea, however it is a boatload of work and is only sustainable if there're enough contributors and maintainers driving it forward. There might be some synergy effects by doing it all in one but then again there're enough variants to make it a lot of work to figure out a common scheme. It's also questionable if there's a benefit of a different approach over `embassy-stm32` to make it worthwhile to start over. I have my serious doubts about that and would note that even `embassy-stm32` is somewhat incomplete in various areas...

burrbull, 2025-07-17:
I think this should be done step by step.
1. Allow to review PRs in all `-hal` crates for all active maintainers from other hal crates. So we can help to each other.
2. Make a list of APIs from different hals (with link) which we consider good or acceptable. And spread them to other hal.
3. Only after these steps we can decide how to reorganize crates:
a. One crate to rule all families.
b. Leave all as it is.
c. Something in between. Crates that collect several similar families. Like F2+F4+F7, G0+C0, H5+U5, etc.

The separate question is should we collect all repos in one from start?
Advantage is we can skip first step and partially third.
Disadvantage: this big work somebody should do and also see "15 competing standards" meme.

usbalbin, 2025-07-17:
Maybe we should summarize what makes this different from embassy's hal?

David-OConnor, 2025-07-17:
Note: As of PAC 0.16, many of the cases of PAC syntax differences between variants have gone away.

I made heavy use of the cfg_if macro, and feature gates in general. Sometimes for whole families, sometimes for specific variants. If you embark up on this quest, I will borrow any good ideas you come up with to simplify this process, and make the code to read and maintain. 

For example, I recently found the cfg_aliases lib. It lets you set up combined feature gates.

astapleton, 2025-07-17:
> [@astapleton](https://github.com/astapleton) That has been suggested before and indeed seems to be one of the most promising approaches, however there's one major caveat: The owned PAC singleton which makes it really hard to have separate HALs sharing driver implementations for individual peripherals via dedicated crates.

Maybe I'm missing something, but why is that an issue? If the driver crate is integrated into the individual PAC and provides it's own extension trait to take ownership of the SPI peripheral, like most stm32 HAL drivers do, what would be the difference?

astapleton, 2025-07-17:
In terms of why not use embassy, I'm not super enthusiastic about the HAL for the processor family being owned by an organization that has its own interests. While I don't have any problems with the embassy organization (and plan to use embassy myself!), I feel like putting the destiny in their hands invites potential conflicts of interest down the line (e.g. maybe they don't want to allow support for a competing runtime). I would much rather it belong to a community that is simply focused on providing a HAL for the processor family.

burrbull, 2025-07-17:
> Maybe I'm missing something, but why is that an issue? If the driver crate is integrated into the individual PAC and provides it's own extension trait to take ownership of the SPI peripheral, like [...]

I think that you are talking about slightly different approaches:
Like https://github.com/stm32-rs/bxcan vs https://github.com/usbalbin/stm32-hrtim/

David-OConnor, 2025-07-17:
On that note: `bxcan`, `fdcan`, `usb`, and `synopsis-usb-otg` are fantastic libraries that I have successfully integrated. They offload most of the work that would otherwise be required in HAL, and I've experienced no integration problems with them, across all relevant STM32 families.

mbrossard, 2025-07-18:
In addition to `embassy-stm32` @ryan-summers mentioned in Matrix https://crates.io/crates/stm32-hal2 (I see its author @David-OConnor is already in the thread).

At RustWeek Unconference, I volunteered to write a document on "Supporting a new SoC" which is now up for review (https://github.com/rust-embedded/embedonomicon/pull/99). As you are thinking about the future of STM32 HALs, I would appreciate if you could check that its content makes sense (or if you have additional suggestions).

David-OConnor, 2025-07-18:
With that in mind, you will find there [the only?] prior art of integrating multiple STM32 families using the stm32-rs PAC libs. For anyone not tracking, there are reconcilable design preferences that will keep these two projects (stm32-hal and stm32yxx-hals) separate. Summarized concisely as the former aimed towards firmware.

The bulk of the integration is done using feature gates, including with `cfg_if`. I don't think this is an ideal pattern, and there may be a better one. If so, I don't know what it is. For peripherals that are highly conserved, the code is clean. Examples include IDWG, RTC, USB (sort of), FD-CAN, USB, and DMA. For most IO, it's a mix (SPI, I2C etc). At the other end, the timer and RCC code is very messy due to feature gates.

burrbull, 2025-07-21:
Alternative way.

We can start unifying from creating some crate which lays in between of PACs and HALs. It will contain different peripheral implementations (similar to `hrtim` crate) hidden under features.

[image]

So HAL maintainer can choose: 
a) use existent common implementation of peripheral
b) use own implementation
c) add new implementation to meta crate

usbalbin, 2025-07-21:
cc @AdinAck

David-OConnor, 2025-07-21:
Would that look like the fdcan and usb libs?

David-OConnor, 2025-07-22:
It is not a viable option, because it would require a complete rewrite. It's not a bad approach, but I'm also not sure if it's better than the PAC approach. It's also unclear to me, for example, which combinations of peripherals on which MCUs are supported (and working well/tested) on Embassy; the docs and examples are a bit sparse there.

So, this is a scenario where maybe you weigh PAC vs Embassy Metapac, and find Embassy Metapac is a bit better. Not good enough; would have to be *much* better to justify the amount of rewrite and testing required.